kpreid/shinysdr #128: Don't use rtl_433 deprecated option

Flupsy, 2019-02-11:
With the latest rtl_433 shinysdr stops working when this mode is selected because it passes `-m 3` to `rtl_433`, and that option just makes rtl_433 spit out a usage message and exit.

See this code from `rtl_433.c`:

    case 'm':
        fprintf(stderr, "sample mode option is deprecated.\n");
        usage(NULL, 0, 1);
        break;

kpreid, 2019-02-12:
I've fixed the options so that it'll run, but now it never decodes anything. Will need to investigate further.

kpreid, 2019-03-17:
It turns out that the lack of decoding was apparently due to a rtl_433 bug fixed 3 days ago: https://github.com/merbanan/rtl_433/pull/1009 https://github.com/merbanan/rtl_433/commit/138996e473a4738d4f4b8f4f5db8665ded77d0e9

ShinySDR as of 040671de51bb6aa5b282207635c99428e18a1663 works with current rtl_433. There is no numbered rtl_433 release to refer to yet.

closed 2019-03-17 by kpreid

mathisono, 2019-03-17:
> fixed 3 days ago: merbanan/rtl_433#1009
> <https://github.com/merbanan/rtl_433/pull/1009> merbanan/rtl_433@138996e

So ? Was there a modification to our code base or just the rtl_433?

Im going to take this opportunity to say that the documentation could
include a little more detail on how to duplicate/configure plugins. Like if
if one wanted to duplicate the aprs plugin for 9600bps and have 2 aprs
modes as options in the ui.

Neet tool this is...!

73 Mathison

kpreid, 2019-03-17:
> So ? Was there a modification to our code base or just the rtl_433?

Sorry, I can see now that wasn't entirely clear.

* Commit ab5b90640ebffe3b46ffcc7cb31ebbfcf421bc74 modified the options ShinySDR passes to rtl_433, which fixes the reported issue.
* Separately, rtl_433 had a bug that made complex-float32 input not decode properly, which is why this issue sat open for a while because I couldn't confirm the modified options were correct (or that it was a problem on rtl_433's end either).

> Im going to take this opportunity to say …

To keep things well organized for everyone's reading, it is much better to file separate issues.

> … that the documentation could include a little more detail on how to duplicate/configure plugins. … if one wanted to duplicate the aprs plugin for 9600bps and have 2 aprs modes as options in the ui.

The documentation is short because ShinySDR's interfaces are not particularly stable, so I don't want to write documentation that becomes obsolete and confusing until things are nailed down better. The best place to start is looking at the existing plugins.

But you don't need to duplicate a plugin to add an option. Plugins can define as many modes as they like (for example, `basic_demod` defines all of AM FM WFM CW USB LSB and shares code between them).

I'll see if I can think of some documentation improvements. For any further matters, please **open a separate issue.**