kpreid/shinysdr #130: Unselective AM

mathisono, 2019-02-16:
Unselective AM is great! When you have enough dynamic range to hear week signals and strong over the noise floor.   Its a great method to monitor ATC radio traffic.. 

I would suspect that AM  will find and acquire a signal, only for baseband reference, then pull the base band chunk into an revolving FM demodulation slots with AGC to rotate around acquired signals.? 

https://kpreid.livejournal.com/52797.html

https://qsl.net/kj6dzb/HAM/sdr%20makin.html

https://qsl.net/kj6dzb/musicflv/OaKCenter%20ATC%20unselective%20recording%202_15_2019.wav

kpreid, 2019-02-16:
Sorry, I'm not quite sure — is there a feature request here?

If you're asking about detecting carriers so as to use non-AM demodulators on them, that is the sort of thing that might happen in the future but it will most likely be derived from FFT power measurements (i.e. looking for the same peaks you see on screen) rather than the unselective AM mechanism.

mathisono, 2019-04-01:
no feature request, just wanted to share some audio from the AM demodulator.

closed 2019-04-01 by kpreid