kpreid/shinysdr #132: get_threshold error when switching to s-mode

FoxxMD, 2019-04-12:
When I try to switch to s-mode all of the gui contols become unresponsive and audio stops, but I can still see the waterfall. Here is the output from the log when I click s-mode:

```
INFO:shinysdr:Flow graph: Rebuilding connections because: receiver a: changed output type
INFO:shinysdr:Flow graph: ...done reconnecting (3.79204750061 ms).
WARNING:shinysdr:Discarding nonexistent state <shinysdr.plugins.mode_s.ModeSDemodulator object at 0x14d030074f90>.demod_method = async

WARNING:shinysdr:Discarding nonexistent state <shinysdr.plugins.mode_s.ModeSDemodulator object at 0x14d030074f90>.squelch_threshold = -93.7095032397

INFO:shinysdr:Flow graph: Rebuilding connections because: receiver a: changed output type
INFO:shinysdr:Flow graph: ...done reconnecting (0.584125518799 ms).
CRITICAL:shinysdr:Unhandled Error

Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 55, in main
react(go)
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/task.py", line 936, in react
_reactor.run()
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 1272, in run
self.mainLoop()
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 1281, in mainLoop
self.runUntilCurrent()
--- <exception caught here> ---
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 902, in runUntilCurrent
call.func(*call.args, **call.kw)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 213, in __call__
self.__handle_value(value)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 112, in __listen_cell
self.__ssi._lookup_or_register(value, self.url)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 284, in _lookup_or_register
registration = _StateStreamObjectRegistration(ssi=self, subscription_context=self.__subscription_context, obj=obj, serial=serial, url=url, refcount=0, send_registration=True)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 81, in __init__
self.__listen_state(initial_value)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 139, in __listen_state
self.__send_references_and_update_refcount(state, False)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 145, in __send_references_and_update_refcount
for k, v in six.iteritems(objs)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 145, in <dictcomp>
for k, v in six.iteritems(objs)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 284, in _lookup_or_register
registration = _StateStreamObjectRegistration(ssi=self, subscription_context=self.__subscription_context, obj=obj, serial=serial, url=url, refcount=0, send_registration=True)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 59, in __init__
initial_value, self.__subscription = obj.subscribe2(subscriber, subscription_context)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 351, in subscribe2
return self.get(), subscription
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 330, in get
value = self.__getter()
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/__init__.py", line 127, in get_decode_threshold
return self.__demod.get_threshold()
exceptions.TypeError: get_threshold() takes exactly 2 arguments (1 given)

CRITICAL:shinysdr:Unhandled Error

Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 55, in main
react(go)
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/task.py", line 936, in react
_reactor.run()
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 1272, in run
self.mainLoop()
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 1281, in mainLoop
self.runUntilCurrent()
--- <exception caught here> ---
File "/usr/local/lib/python2.7/dist-packages/Twisted-19.2.0rc2-py2.7-linux-x86_64.egg/twisted/internet/base.py", line 902, in runUntilCurrent
call.func(*call.args, **call.kw)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/persistence.py", line 114, in __write_immediately
current_state = self.__pcd.get() # note: may raise if getters are broken
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/persistence.py", line 145, in get
return self.__root.state_to_json(subscriber=self.__add_subscription)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 786, in state_to_json
state[key] = cell.get_state(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 228, in get_state
return self.get().state_to_json(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 786, in state_to_json
state[key] = cell.get_state(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 228, in get_state
return self.get().state_to_json(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 786, in state_to_json
state[key] = cell.get_state(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 228, in get_state
return self.get().state_to_json(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 786, in state_to_json
state[key] = cell.get_state(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 228, in get_state
return self.get().state_to_json(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 786, in state_to_json
state[key] = cell.get_state(subscriber=subscriber)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 226, in get_state
subscriber(self.subscribe2)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/persistence.py", line 155, in __add_subscription
_value, subscription = subscribe_fn(self.__do_callback, self.__subscription_context)
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 351, in subscribe2
return self.get(), subscription
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/values.py", line 330, in get
value = self.__getter()
File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/__init__.py", line 127, in get_decode_threshold
return self.__demod.get_threshold()
exceptions.TypeError: get_threshold() takes exactly 2 arguments (1 given)

INFO:shinysdr:(-40 1.52138900) No handler for message type 24 from 420c99
INFO:shinysdr:(-40 2.02204550) Type 4 (short surveillance altitude reply) from f64edf at 85500ft (AIRBORNE ALERT)
INFO:shinysdr:(-41 4.07859050) No handler for message type 24 from 45911c
INFO:shinysdr:(-42 4.20955600) Type 4 (short surveillance altitude reply) from 95f01e at 7275ft (SPI)
INFO:shinysdr:(-40 6.70005050) No handler for message type 24 from d5c861
INFO:shinysdr:(-42 7.18729000) No handler for message type 24 from 2a57c7
```

I am already tuned to the ads-b frequency and as you can see I get telemetry output to the log but not to the interface. A restart of the application is required to get functionality back and state is not saved.

kpreid, 2019-04-12:
You have an old version of gr-air-modes installed. Make sure it's newer than this change: https://github.com/bistromath/gr-air-modes/commit/719d52b6dd6a88a809245ea26a759263426548d4

I'll see about improving the handling of this case. (At least catching it up front and giving a nice error; unfortunately, it's tricky to keep the UI running in the case of "we tried to read this value for the very first time and got an exception instead".)

FoxxMD, 2019-04-12:
Thanks! I'm running this from a docker, didn't realize I had an old version. I will try to update

jeffersonjhunt, 2019-04-12:
I'll update this today and push out a new version of the docker container.

FoxxMD, 2019-04-12:
thanks @jeffersonjhunt !

jeffersonjhunt, 2019-04-12:
This should be fixed via version 1.1.0 of my Docker container. Please update.

FoxxMD, 2019-04-12:
That did it! Thank you. Mode S can be selected now, I can see decode threshold and message/sec decoded 🎉

FoxxMD, 2019-04-12:
Additional question though -- I can see messages/decoded and see ADB-S data in the log but the Telemetry window rarely registers anything. @kpreid is there some kind of filter for what kind of data that window displays? I didn't see anything in the manual...

closed 2019-04-12 by FoxxMD

kpreid, 2019-04-13:
@FoxxMD Yes, there is an interestingness filter: it is defined as `Aircraft.is_interesting` at https://github.com/kpreid/shinysdr/blob/ffd2c43e5435eff3748dd352e963c94d69af7c8f/shinysdr/plugins/mode_s/__init__.py#L252 [note: pinned to a version to ensure the line number is stable]. The filter requires that there be either ID or position information.

When I originally wrote the filter I found that this was necessary to not have a large number of uninteresting records. (I now think that some of those were probably bad decodes and the threshold could be used to fix that.) As the comment notes, I'd like to have better UI (client-side filtering) before loosening that filter.