kpreid/shinysdr #135: Must have at least one RF device

KC9S, 2019-06-24:
I have an issue similar to #112 but I did not see a solution there.  I'm running three RTL-SDR devices on my RPi 3.  The first two use rtl_tcp to feed other programs.  I would like to use with shinysdr with the third device.  Each device has its own serial number.  I believe the issue is with librtlsdr conflicting with gr-osmosdr.  The following is my output when I try to run shinysdr:

````
WARNING:shinysdr:No network service defined!
INFO:shinysdr:Constructing...
CRITICAL:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/task.py", line 908, in react
    finished = main(_reactor, *argv)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 50, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1613, in unwindGenerator
    return _cancellableInlineCallbacks(gen)
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1529, in _cancellableInlineCallbacks
    _inlineCallbacks(None, g, status)
--- <exception caught here> ---
  File "/home/pi/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1418, in _inlineCallbacks
    result = g.send(result)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 101, in _main_async
    app = config_obj._create_app()
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/config.py", line 91, in _create_app
    features=self.features._get_all())
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/session.py", line 37, in __init__
    features=features)
  File "/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/top.py", line 74, in __init__
    raise ValueError('Must have at least one RF device')
exceptions.ValueError: Must have at least one RF device

INFO:shinysdr:Main loop terminated.
````

I'm a novice when it comes to this and am hoping to learn.  Please help!  None of these web based controls hold a candle to shinysdr in my opinion.

kpreid, 2019-06-25:
"Must have at least one RF device" does not mean that no devices were found when you started ShinySDR. It means that **your `config-dir/config.py` file does not define any devices.**  it looks like your configuration file might be missing other parts, because I notice it's also reporting `No network service defined!` — that is, the configuration file is missing `config.serve_web(...)`.

You could edit the configuration to fix whatever's wrong with it, or you could create a new one with `shinysdr --create`. Maybe your config.py is just empty?

(In order to select a device by serial number, create the device like `OsmoSDRDevice('rtl=yourserialnumberhere')`.)

I'll look into improving some of these error messages to communicate what needs to be done and/or what the source of the problem is, not just that a problem exists.

> I believe the issue is with librtlsdr conflicting with gr-osmosdr.

If you mean that you had incompatible library versions installed, then yes, that would explain the problem. The config file generator makes the decision as to whether to include `OsmoSDRDevice` based on whether importing the `osmosdr` Python module succeeds at that time.

KC9S, 2019-06-25:
I think we can chalk this one up to a sloppy install.  I had installed the config directory into my source.  Thus the proper config.py file was not installed.  I’m still not receiving any audio in Firefox despite allowing auto play.  I’m not sure what is going on there.  But for this particular issue I believe I will close it.  Nice work.  I think I’ll really enjoy shinysdr!

closed 2019-06-25 by KC9S

kpreid, 2019-06-29:
6aea642a8365b28f993861098f1f35cfa6d5ade0 and f358b83bccdcecf54c4dad1bc9d27eeac1aa1248 should make the error messages clearer, with less extraneous text.