kpreid/shinysdr #138: Instructions not current?

ofadam, 2020-01-14:
I was following along with the instructions here:

https://shinysdr.switchb.org/manual/installation

And noticed that the "fetch-js-deps" portion of the installation doesn't seem to be accurate any longer. Are there other parts of this document that also need updated?

kpreid, 2020-01-15:
Thanks for catching this! The manual has already been updated upstream (<https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/manual/installation.html>) but pushing it to https://shinysdr.switchb.org/ is a manual process which I forgot to do. If you want to follow the latest copy, for now you can download and view that HTML or read it in source form.

kpreid, 2020-01-18:
https://shinysdr.switchb.org/manual/installation has been updated. Please let me know if you have any other trouble!

closed 2020-01-18 by kpreid