kpreid/shinysdr #143: shinysdr --create dir-name

forteuropa, 2020-05-04:
Hello I'm having problems with creating config directory, when i add command shinysdr --create /configSDR/ i get this: 
[fort@localhost shinysdr]$ shinysdr --create home
Traceback (most recent call last):
  File "/usr/local/bin/shinysdr", line 11, in <module>
    load_entry_point('ShinySDR==0.0.0', 'console_scripts', 'shinysdr')()
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2855, in load_entry_point
    return ep.load()
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2446, in load
    return self.resolve()
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2452, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.7/site-packages/shinysdr/main.py", line 37, in <module>
    from shinysdr.i.config import Config, ConfigException, write_default_config, execute_config, print_config_exception
  File "/usr/local/lib/python3.7/site-packages/shinysdr/i/config.py", line 42, in <module>
    from shinysdr.i.db import DatabaseModel, database_from_csv, databases_from_directory
  File "/usr/local/lib/python3.7/site-packages/shinysdr/i/db.py", line 35, in <module>
    from shinysdr.i.network.base import ElementRenderingResource, template_filepath
  File "/usr/local/lib/python3.7/site-packages/shinysdr/i/network/base.py", line 38, in <module>
    from shinysdr.i.roots import IEntryPoint
  File "/usr/local/lib/python3.7/site-packages/shinysdr/i/roots.py", line 27, in <module>
    from shinysdr.values import CellDict, CollectionState
  File "/usr/local/lib/python3.7/site-packages/shinysdr/values.py", line 36, in <module>
    from gnuradio import gr
ModuleNotFoundError: No module named 'gnuradio'

P.S I am using python 2.7 by default. 
Any suggestions ?

kpreid, 2020-05-04:
From the module path, you're running Python 3. Check if your GNU Radio installation is compatible — you need to either use GNU Radio 3.8 or later for Python 3 compatibility, or run ShinySDR with Python 2.7 as GR 3.7 or earlier only works with Python 2.

(Note: Unfortunately, because the package I use myself is lagging, I haven't gotten to testing ShinySDR with GR 3.8 yet, so there might be problems. I've worked to ensure ShinySDR runs on Python 2 and 3 but only the non-GR-using parts have been actually _tested._ But, that's not the problem you're seeing.)

forteuropa, 2020-05-04:
My GNU Radio version is 3.7.13.5 and i set my comp to run programs with python2.7 by default:
GNU Radio Companion 3.7.13.5

This program is part of GNU Radio
GRC comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it.

[fort@localhost shinysdr]$ python --version 
Python 2.7.17

kpreid, 2020-05-04:
The version being used by the shinysdr command is 3.7 as indicated by the paths: 

    ...
    File "/usr/local/lib/python3.7/site-packages/shinysdr/i/config.py"
    ...

So, you need to reinstall shinysdr using your python 2.7.17 instead of your python 3.7.

forteuropa, 2020-05-04:
After i remove everything, I started fresh install. I used comand $python2.7 setup.py build and python2.7 setup.py install. Thats the full log I get : 

fort@localhost shiny]$ git clone https://github.com/kpreid/shinysdr.git
Cloning into 'shinysdr'...
remote: Enumerating objects: 144, done.
remote: Counting objects: 100% (144/144), done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 14734 (delta 78), reused 88 (delta 47), pack-reused 14590
Receiving objects: 100% (14734/14734), 6.17 MiB | 5.06 MiB/s, done.
Resolving deltas: 100% (10319/10319), done.
[fort@localhost shiny]$ cd shinysdr/
[fort@localhost shinysdr]$ python2.7 setup.py build
running build
running build_py
running fetch_deps
running git_init
Initializing submodules...
Submodule 'shinysdr/deps/aprs-symbol-index' (https://github.com/hessu/aprs-symbol-index) registered for path 'shinysdr/deps/aprs-symbol-index'
Submodule 'shinysdr/deps/aprs-symbols' (https://github.com/hessu/aprs-symbols) registered for path 'shinysdr/deps/aprs-symbols'
Submodule 'shinysdr/deps/jasmine' (https://github.com/jasmine/jasmine) registered for path 'shinysdr/deps/jasmine'
Submodule 'shinysdr/deps/measviz' (https://github.com/kpreid/measviz/) registered for path 'shinysdr/deps/measviz'
Cloning into '/home/fort/sdr/shiny/shinysdr/shinysdr/deps/aprs-symbol-index'...
Cloning into '/home/fort/sdr/shiny/shinysdr/shinysdr/deps/aprs-symbols'...
Cloning into '/home/fort/sdr/shiny/shinysdr/shinysdr/deps/jasmine'...
Cloning into '/home/fort/sdr/shiny/shinysdr/shinysdr/deps/measviz'...
Submodule path 'shinysdr/deps/aprs-symbol-index': checked out '68a273408fa12ea01a9fc50b8bc3e32c31f0f508'
Submodule path 'shinysdr/deps/aprs-symbols': checked out 'ec24de8c8c66674cbf030353a80a3b5111b7e93d'
Submodule path 'shinysdr/deps/jasmine': checked out '6816bc4252fe40ea2cf6a16b4aac7ddce6922095'
Submodule path 'shinysdr/deps/measviz': checked out '86cbcd498926790c3823bee5dd6ab7d40ac664a9'
running retrieve_assets
downloading https://raw.githubusercontent.com/requirejs/text/646db27aaf2236cea92ac4107f32cbe5ae7a8d3a/text.js to shinysdr/deps/text.js
downloading http://requirejs.org/docs/release/2.1.22/comments/require.js to shinysdr/deps/require.js
creating build
creating build/lib
creating build/lib/shinysdr
copying shinysdr/__init__.py -> build/lib/shinysdr
copying shinysdr/devices.py -> build/lib/shinysdr
copying shinysdr/test_values.py -> build/lib/shinysdr
copying shinysdr/testutil.py -> build/lib/shinysdr
copying shinysdr/telemetry.py -> build/lib/shinysdr
copying shinysdr/test_twisted_ext.py -> build/lib/shinysdr
copying shinysdr/math.py -> build/lib/shinysdr
copying shinysdr/test_db_import.py -> build/lib/shinysdr
copying shinysdr/test_grc.py -> build/lib/shinysdr
copying shinysdr/test_types.py -> build/lib/shinysdr
copying shinysdr/types.py -> build/lib/shinysdr
copying shinysdr/filters.py -> build/lib/shinysdr
copying shinysdr/signals.py -> build/lib/shinysdr
copying shinysdr/units.py -> build/lib/shinysdr
copying shinysdr/test_telemetry.py -> build/lib/shinysdr
copying shinysdr/test_filters.py -> build/lib/shinysdr
copying shinysdr/interfaces.py -> build/lib/shinysdr
copying shinysdr/values.py -> build/lib/shinysdr
copying shinysdr/twisted_ext.py -> build/lib/shinysdr
copying shinysdr/test_devices.py -> build/lib/shinysdr
copying shinysdr/test_signals.py -> build/lib/shinysdr
copying shinysdr/main.py -> build/lib/shinysdr
creating build/lib/shinysdr/test_manually
copying shinysdr/test_manually/__init__.py -> build/lib/shinysdr/test_manually
copying shinysdr/test_manually/aprs_parser.py -> build/lib/shinysdr/test_manually
copying shinysdr/test_manually/channel_filter_benchmark.py -> build/lib/shinysdr/test_manually
creating build/lib/shinysdr/plugins
copying shinysdr/plugins/__init__.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/rtl_433.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/basic_demod.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/limesdr.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/test_simulate.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/test_aprs.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/test_psk31.py -> build/lib/shinysdr/plugins
copying shinysdr/plugins/test_osmosdr.py -> build/lib/shinysdr/plugins
[... 876 lines cut ...]
Using /usr/lib/python2.7/site-packages
Searching for pycparser==2.20
Best match: pycparser 2.20
Processing pycparser-2.20-py2.7.egg
pycparser 2.20 is already the active version in easy-install.pth

Using /usr/lib/python2.7/site-packages/pycparser-2.20-py2.7.egg
Finished processing dependencies for ShinySDR==0.0.0
[fort@localhost shinysdr]$ shinysdr --create /setupSDR/
Traceback (most recent call last):
  File "/usr/local/bin/shinysdr", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3254, in <module>
    @_call_aside
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3238, in _call_aside
    f(*args, **kwargs)
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3267, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 584, in _build_master
    ws.require(__requires__)
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/fort/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'ShinySDR==0.0.0' distribution was not found and is required by the application

kpreid, 2020-05-04:
It looks like the `shinysdr` command is still running python 3.7. Try deleting `/usr/bin/shinysdr` and reinstalling.

forteuropa, 2020-05-04:
Ok thank you so much,  now i was able to create config dir, but when i execute $shinysdr ./config-myConfig-dir i get error. Here is the output 

[fort@localhost shinysdr]$ shinysdr --create setupSDR
INFO: Audio source arch: alsa
INFO:shinysdr:Created default configuration at: setupSDR
INFO:shinysdr:Main loop terminated.
[fort@localhost shinysdr]$ ls
AUTHORS  COPYING  lint.sh      pylintrc   setup.py  shinysdr
build    dist     MANIFEST.in  README.md  setupSDR  ShinySDR.egg-info
[fort@localhost shinysdr]$ shinysdr ./setupSDR
CRITICAL:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/Twisted-20.3.0rc1-py2.7-linux-x86_64.egg/twisted/internet/task.py", line 909, in react
    finished = main(_reactor, *argv)
  File "/usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 51, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/usr/lib/python2.7/site-packages/Twisted-20.3.0rc1-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1613, in unwindGenerator
    return _cancellableInlineCallbacks(gen)
  File "/usr/lib/python2.7/site-packages/Twisted-20.3.0rc1-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1529, in _cancellableInlineCallbacks
    _inlineCallbacks(None, g, status)
--- <exception caught here> ---
  File "/usr/lib/python2.7/site-packages/Twisted-20.3.0rc1-py2.7-linux-x86_64.egg/twisted/internet/defer.py", line 1418, in _inlineCallbacks
    result = g.send(result)
  File "/usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 104, in _main_async
    execute_config(config_obj, args.config_path)
  File "/usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/config.py", line 354, in execute_config
    _execute_config_file(config_obj, os.path.join(config_file_or_directory, 'config.py'))
  File "/usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/config.py", line 375, in _execute_config_file
    six.exec_(code, env)
  File "/usr/lib/python2.7/site-packages/six.py", line 709, in exec_
    exec("""exec _code_ in _globs_, _locs_""")
  File "<string>", line 1, in <module>
    
  File "./setupSDR/config.py", line 13, in <module>
    config.devices.add(u'osmo', OsmoSDRDevice(''))
  File "/usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/osmosdr.py", line 227, in OsmoSDRDevice
    raise Exception('OsmoSDRDevice: gr-osmosdr Python bindings not found; cannot create device')
exceptions.Exception: OsmoSDRDevice: gr-osmosdr Python bindings not found; cannot create device

INFO:shinysdr:Main loop terminated.

kpreid, 2020-05-04:
As the message says, you are missing the Python bindings for gr-osmosdr. Make sure you have gr-osmosdr installed. I've heard there was some delay in updating gr-osmosdr for GNU Radio 3.8 — you might need to get a package from somewhere other than your distribution; I don't know what the situation on Fedora is.

A quick direct test without using shinysdr is to run `python -c 'import osmosdr'` (using the same python version, of course) — if gr-osmosdr with python bindings is correctly installed it should complete without any message, and if it is not you should see an `ImportError`.

forteuropa, 2020-05-06:
Thanks for your replay, installing gr-osmosdr server was able to start, but as soon i enter the link localhost:8100/........./ i get segmentation fault(core dumped) error. Here is the output: 
fort@localhost shinysdr]$ shinysdr /home/fort/sdr/shiny/shinysdr/setupSDR/
gr-osmosdr v0.1.x (0.1.5git) gnuradio 3.7.13.5
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 9.1.1 20190605 (Red Hat 9.1.1-2); Boost_106900; UHD_3.14.1.0
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
INFO: Audio source arch: alsa
INFO:shinysdr:Loading mode plugins...
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/elecraft/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/wspr/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/psk31/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:Mode RTTY unavailable
No module named radioteletype.demodulators
INFO:shinysdr:Mode MODE-S unavailable
No module named air_modes
INFO:shinysdr:Mode APRS unavailable
Expected `multimon-ng -h; exit 0` to give output containing 'vailable demodulators:', but the actual output was:
/bin/sh: multimon-ng: command not found

INFO:shinysdr:Mode WSPR unavailable
wsprd not found.
INFO:shinysdr:Mode PSK31 unavailable
No module named radioteletype.demodulators
INFO:shinysdr:Mode 433 unavailable
Expected `rtl_433 -r /dev/null` to succeed but it could not be started.
INFO:shinysdr:Mode DSD unavailable
dsd.dsd_block_ff not found.
INFO:shinysdr:...done mode plugins.
INFO:shinysdr:Constructing...
INFO: Audio source arch: alsa
INFO:shinysdr:Flow graph: Switching RF device to audio
INFO: Audio source arch: alsa
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO: Audio source arch: alsa
INFO:shinysdr:Flow graph: ...done reconnecting (13.9331817627 ms).
INFO:shinysdr:Restoring state...
WARNING:shinysdr:Discarding nonexistent state <shinysdr.i.session.AppRoot object at 0x7f60b3691650>.source_name = osmo
INFO:shinysdr:Starting web server...
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/elecraft/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/wspr/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/psk31/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:SiteWithDefaultHeaders starting on 8100
INFO:shinysdr:Starting factory <shinysdr.i.network.base.SiteWithDefaultHeaders instance at 0x7f60b2ac45a0>
INFO:shinysdr:WebSocketDispatcherProtocol (FactoryWithArgs) (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7f60b2ac47d0>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f60b2ac4820>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/ow13fJLNEsx3MeGgsK78ZQ/
INFO:shinysdr:"127.0.0.1" - - [06/May/2020:13:24:39 +0000] "GET /ow13fJLNEsx3MeGgsK78ZQ/ HTTP/1.1" 200 4366 "-" "Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:75.0) Gecko/20100101 Firefox/75.0"
INFO:shinysdr:"127.0.0.1" - - [06/May/2020:13:24:39 +0000] "GET /client/ui.css HTTP/1.1" 200 22420 "-" "Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:75.0) Gecko/20100101 Firefox/75.0"
[... cut ...]
INFO:shinysdr:"127.0.0.1" - - [06/May/2020:13:24:40 +0000] "GET /client/plugins/shinysdr.plugins.aprs/symbols/aprs-symbols-24-2%402x.png HTTP/1.1" 200 24929 "-" "Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:75.0) Gecko/20100101 Firefox/75.0"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:"127.0.0.1" - - [06/May/2020:13:24:40 +0000] "GET /ow13fJLNEsx3MeGgsK78ZQ/dbs/ HTTP/1.1" 200 1695 "-" "Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:75.0) Gecko/20100101 Firefox/75.0"
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to /ow13fJLNEsx3MeGgsK78ZQ/radio
INFO: Audio source arch: alsa
Segmentation fault (core dumped)

kpreid, 2020-05-07:
This is almost certainly a bug in GNU Radio or other libraries loaded by ShinySDR, but there might be a workaround. In order to give any further advice I will need a stack trace extracted from that core dump.

forteuropa, 2020-05-08:
Alright yesterday i create ubuntu19  VM on my machine and install shinysdr with gnuradio 3.7 without any problems, but because its VM i was having some serious lags, so i decided to fully migrate to Ubuntu. I installed ubuntu 20  but when i installed GNUradio and tried to run shinysdr i got error: ImportError: No module named gnuradio

Again i installed Shinysdr with python 2.7 , but this time my GNUradio was 3.8 version,  I guess thats why when i type python >>> import gnuradio , its gives me error no module, but when i type:       python3 >>> import gnuradio , there is no longer error. 

So my question is is it possible to install Shinysdr with GNUradio 3.8 or latter, because  I am having difficult installing GNUradio 3.7 or earlier ?

kpreid, 2020-05-08:
I have not yet tested ShinySDR with GNU Radio 3.8. I don't know of any reason it won't work, and I certainly want to fix it if it doesn't, but I can't make any guarantees, and it's highly likely there will be _some_ problem.

joshspivey, 2020-06-24:
Yah 3.8 is a major upgrade and it won’t work. I wanted to write some TSCM modules and signal intelligence modules for it but thing I’m going to scrap that idea because it can’t work w 3.8