kpreid/shinysdr #144: Plugins not working when running unprivileged

fmoessbauer, 2020-06-02:
Hi, with the recent change that shinysdr has to be run as a non-privileged user, decoding plugins do not work anymore due to wrong permissions of the cache directory.

**Steps to reproduce:**

- build shiny
- install shiny
- change to non-privileged user (e.g. sdr)
- run shiny

```
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Mode RTTY unavailable
No module named radioteletype_swig
INFO:shinysdr:Mode MODE-S unavailable
No module named air_modes_swig
INFO:shinysdr:Mode PSK31 unavailable
No module named radioteletype_swig
INFO:shinysdr:Mode DSD unavailable
dsd.dsd_block_ff not found.
```

After giving user sdr write access to `/usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg`, all plugins can be loaded and work as expected.

Is it possible to change the cache location? Changing the access rights under `/usr/local` is not good practice IMO.

kpreid, 2020-06-02:
> Hi, with the recent change that shinysdr has to be run as a non-privileged user, decoding plugins do not work anymore due to wrong permissions of the cache directory.

"Unable to write to plugin cache" is harmless in that it merely results in a slower startup, and in fact the log you quoted shows that the plugins *were* succesfully loaded. For example, the log containing "Mode RTTY unavailable" demonstrates that the RTTY plugin **was successfully loaded** — it ran its code to check if it can actually operate, and found that it couldn't.

There must be something else different that allowed the modes to work. "No module named ...\_swig" generally means that you have GNU Radio Python libraries but not the matching C++ libraries on your loading paths, which is a common problem when you have an incomplete installation or are using a different Python environment than GR was installed under.

fmoessbauer, 2020-06-02:
Hi @kpreid , thanks for the explanation. Actually my first guess was not right. The plugins were not loaded correctly, but the reason was that the swig libraries where missing (I'm working on a minimal footprint docker container with all plugins). A simple `apt install swig` before building the plugins (e.g. gr-air-modes) helped to solve the issue.
In this case, the logs were just misleading.

Maybe, it would be better to either hide the plugin cache error if possible, or display a message that this is a known limitation.

Anyways, sorry for the noise. Shinysdr is a really cool piece of software! 73, DL1FMP

closed 2020-06-02 by fmoessbauer