kpreid/shinysdr #148: No audio

FMC-gclifford, 2020-08-27:
Hi All,

New user here (great application).

I see the audio bar jumping around but i'm not getting any volume on my clients.  I've tried Chrome, Firefox and IE and none appear to have any audio.

I'm trying to just listen to Broadcast FM to get started as a test.

[image]

Am I missing something obvious?

Nothing obvious in the logs

-----LOGS---

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
INFO:shinysdr:Loading mode plugins...
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/elecraft/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/psk31/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/wspr/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:Mode RTTY unavailable
No module named radioteletype.demodulators
INFO:shinysdr:Mode MODE-S unavailable
No module named air_modes
INFO:shinysdr:Mode APRS unavailable
Expected `multimon-ng -h; exit 0` to give output containing 'vailable demodulators:', but the actual output was:
/bin/sh: 1: multimon-ng: not found

INFO:shinysdr:Mode WSPR unavailable
wsprd not found.
INFO:shinysdr:Mode PSK31 unavailable
No module named radioteletype.demodulators
INFO:shinysdr:Mode 433 unavailable
Expected `rtl_433 -r /dev/null` to succeed but it could not be started.
INFO:shinysdr:Mode DSD unavailable
dsd.dsd_block_ff not found.
INFO:shinysdr:...done mode plugins.
INFO:shinysdr:Constructing...
INFO:shinysdr:Flow graph: Switching RF device to sim
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
INFO:shinysdr:Flow graph: ...done reconnecting (40.5390262604 ms).
INFO:shinysdr:Restoring state...
INFO:shinysdr:Flow graph: Switching RF device to osmo
INFO:shinysdr:Flow graph: Rebuilding connections because: switched device
INFO:shinysdr:Flow graph: ...done reconnecting (23.7758159637 ms).
INFO:shinysdr:Flow graph: Rebuilding connections because: added receiver b
INFO:shinysdr:Flow graph: ...done reconnecting (2.38919258118 ms).
INFO:shinysdr:Starting web server...
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/aprs/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/elecraft/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/mode_s/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/psk31/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/wspr/dropin.cache: error number 13
INFO:shinysdr:Unable to write to plugin cache /usr/local/lib/python2.7/dist-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/rtty/dropin.cache: error number 13
INFO:shinysdr:SiteWithDefaultHeaders starting on 8100
INFO:shinysdr:Starting factory <shinysdr.i.network.base.SiteWithDefaultHeaders instance at 0x681c8c60>
INFO:shinysdr:WebSocketDispatcherProtocol (FactoryWithArgs) (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x681b97d8>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x681c86c0>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/U_69g4LaS14fncQ-hjQhOw/
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to /U_69g4LaS14fncQ-hjQhOw/radio
Allocating 15 zero-copy buffers
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to /U_69g4LaS14fncQ-hjQhOw/audio-stream?rate=48000
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio callback
Allocating 15 zero-copy buffers
INFO:shinysdr:Flow graph: ...done reconnecting (1140.4531002 ms).
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOINFO:shinysdr:Starting RFC 6455 conversation
OINFO:shinysdr:Stream connection to /U_69g4LaS14fncQ-hjQhOw/audio-stream?rate=48000
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio callback
Allocating 15 zero-copy buffers
INFO:shinysdr:Flow graph: ...done reconnecting (1043.04099083 ms).
INFO:shinysdr:Starting RFC 6455 conversation
INFO:shinysdr:Stream connection to /U_69g4LaS14fncQ-hjQhOw/radio
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

kpreid, 2020-08-27:
Please provide the contents of the browser's console log (omitting any stack traces) so I can see what the client audio status is.

FMC-gclifford, 2020-08-27:
interesting it seems to be working now - looks like it takes a few reloads of the website before it starts working?

kpreid, 2020-08-27:
That's probably #125 then!

FMC-gclifford, 2020-08-27:
I didn't see a definitive fix.  is it still being worked on?

kpreid, 2020-08-27:
It's not being _actively_ worked on but I intend to fix it at some point. The fix will require new UI, a "play button" for the audio. (Which I think is a good idea anyway.)

ThisIsSet-L, 2020-10-04:
This may be an issue with Autoplay Policy in the Web Audio API

Browsers since 2019 have been blocking audio played without an event (button click, for example)

Literally the easiest and best fix is to load the app, and have it in a stopped/standby state, requiring the user to click a play button to start the session.

https://developer.mozilla.org/en-US/docs/Web/API/Web_Audio_API/Best_practices#Autoplay_policy

redge76, 2020-11-27:
You can use "site settings" to allow sound to auto-play by default on you shiny instance. Of course this will change the behaviour only on the system you are currently logged on. 
Click on the icon on the left of the URL on you browser. Select "site settings" -> "sound" and set "allow"
reference:
- https://support.google.com/chrome/answer/114662