kpreid/shinysdr #149: request for clarification attaching an IQ based SDR with rigctl ?

mathisono, 2020-10-18:
Im lost the server will not pass this line: 

config.devices.add(u'audio',name='RSHFiq', AudioDevice(rx_device='Built-in Input',sample_rate=48000,channel_mapping='IQ'))

or 

config.devices.add(u'audio',name='RSHFiq', AudioDevice(rx_device='plughw:0,0',sample_rate=48000,channel_mapping='IQ'))

or 

addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))
----------------------------------------------------

I know that this works:

d = hamlib.connect_to_rig(config.reactor, options=['-m', '2519', '-r', '/dev/ttyUSB0', '-s', '38400'])

--------------------------------

what's the proper call to attach a iq radio?

kpreid, 2020-10-18:
```
config.devices.add(u'audio',name='RSHFiq', AudioDevice(rx_device='Built-in Input',sample_rate=48000,channel_mapping='IQ'))
```

This should work — without frequency control since there's no rig device, of course.

```
addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))
```

There are some typos here (no `d.` in `d.addCallback` and a missing quote in `'IQ'`), and I don't know if it's what you were actually trying or whether you copied it into the issue imprecisely.

Let's start with the simpler case of audio only. Please make sure you share **exactly** the text of the config file (you can leave out your `config.serve_web` call for privacy) and the **full** output of ShinySDR when you try to use that config file.

mathisono, 2020-10-18:
It should add config.devices.add.plugin rigctl control for its VFO and IQ audio input! 

-------------------------------
# -*- coding: utf-8 -*-

# This is a ShinySDR configuration file. For more information about what can
# be put here, read the manual section on it, available from the running
# ShinySDR server at: http://localhost:8100/manual/configuration

from shinysdr.devices import AudioDevice
from shinysdr.plugins.osmosdr import OsmoSDRDevice
from shinysdr.devices import PositionedDevice
#from shinysdr.plugins.simulate import SimulatedDevice
import shinysdr.plugins.hamlib as hamlib

#FRZ DB
config.databases.add_writable_database('/home/w6bb/shinysdr/conFiG/cm87UV.csv')

#####  RADIO SETUP######
config.devices.add(u'osmo1', OsmoSDRDevice('hackrf=4f4397',name='hackrf--(APN-450)'),PositionedDevice(37.90642, -122.35015))
config.devices.add(u'osmo2', OsmoSDRDevice('rtl=0',name='RTL.comV3--(LNAA--DishElement'),PositionedDevice(37.90642, -122.35015))
#config.devices.add(u'osmo1', OsmoSDRDevice('rtl=1',name='Generic-Dish--Element))))'),PositionedDevice(37.90642, -122.35015))
###(APN-450)
##RTL.comV3-LNAA--EDfongelement))))
##name=(preAMP-->discone)

##############################################RShfIQ####
#config.devices.add(u'audio', AudioDevice(rx_device=''))
#sudo grig -m 2519 -r /dev/ttyUSB0 -s 38400

addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))

##hamlib.connect_to_rig(config.reactor, options=['-m', '2519', '-r', '/dev/ttyUSB0', '-s', '38400'])
##config.devices.add(u'audio',name='RSHFiq', AudioDevice(rx_device='Built-in Input',sample_rate=48000,channel_mapping='IQ'))

##addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))
config.wait_for(5)

#config.devices.add(u'audio',name='RSHFiq', AudioDevice(rx_device='plughw:0,0',sample_rate=48000,channel_mapping='IQ'))
##d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))
#d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))

#

##################################

--------------------------------

output
------------------
sudo shinysdr ~/shinysdr/conFiG
CRITICAL:shinysdr:main function encountered error
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/task.py", line 908, in react
    finished = main(_reactor, *argv)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 50, in go
    return _main_async(reactor, argv, _abort_for_test)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1613, in unwindGenerator
    return _cancellableInlineCallbacks(gen)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1529, in _cancellableInlineCallbacks
    _inlineCallbacks(None, g, status)
--- <exception caught here> ---
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 1418, in _inlineCallbacks
    result = g.send(result)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 98, in _main_async
    execute_config(config_obj, args.config_path)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/config.py", line 315, in execute_config
    _execute_config_file(config_obj, os.path.join(config_file_or_directory, 'config.py'))
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/config.py", line 335, in _execute_config_file
    code = compile(f.read(), path, 'exec')
exceptions.SyntaxError: EOL while scanning string literal (config.py, line 28)

INFO:shinysdr:Main loop terminated.

mathisono, 2020-10-18:
It's not a single sdr server!

kpreid, 2020-10-19:
Tip: When posting code sections on GitHub, place "\`\`\`" (three backticks) at the beginning and the end, so that they're formatted properly and not as miscellanous headings.

The error message is`SyntaxError: EOL while scanning string literal (config.py, line 28)`. It tells you line 28. Looking at line 28, we see

```
addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping=IQ')))
```

There is the syntax error it is referring to: you have `channel_mapping=IQ'` instead of `channel_mapping='IQ'`.

But with that fixed, there are still problems: `addCallback` is not a global function; you have to call it on the deferred returned from `hamlib.connect_to_rig`, like so:

```
d = hamlib.connect_to_rig(config.reactor, options=['-m', '2519', '-r', '/dev/ttyUSB0', '-s', '38400'])
d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping='IQ')))
```

That looks like it should work, but I haven't got your hardware so of course I can't test it.

mathisono, 2020-10-19:
pardon the formatting errors!!! 

When I isolated the device in the server. removing the RTL and hackRF, It fails. Looks to be mad "ERROR:shinysdr:ConfigException: No devices have been configured using config.devices.add(...)." 

-----------------------------------------
w6bb@W6BB-IO1:~$ sudo shinysdr ~/shinysdr/conFiG
[sudo] password for w6bb:
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

INFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f7e57c9a0f0>
INFO:shinysdr:[unfiltered]
ERROR:shinysdr:An error occurred while executing the ShinySDR configuration file:
ERROR:shinysdr:  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 99, in _main_async
ERROR:shinysdr:    yield config_obj._wait_and_validate()
ERROR:shinysdr:ConfigException: No devices have been configured using config.devices.add(...).
INFO:shinysdr:Stopping factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f7e57c9a0f0>
INFO:shinysdr:Main loop terminated.
CRITICAL:shinysdr:Unhandled error in Deferred:
CRITICAL:shinysdr:
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 255, in _connect_to_device
    port=actual_port)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 137, in connect_to_rigctld
    proxy_ctor=_HamlibRig)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 168, in _connect_to_daemon
    protocol = yield connected
ConnectError: An error occurred while connecting: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionLos            t'>: Connection to the other side was lost in a non-clean fashion: Connection lost.
].

------------------------------------------

kpreid, 2020-10-19:
Oh, right, you also need to specify that there are things to wait for. End the config file with
```
config.wait_for(d)
```

By the way, you should not be running shinysdr as root with `sudo`. Network servers running as root are unnecessarily dangerous.

mathisono, 2020-10-19:
I haven't fixed the permissions for ttyUSB0  so  im using root for testing.

its still producing errors, having trouble passing frames?

-------------------------
INFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7fa2c0216410>
INFO:shinysdr:[unfiltered]
ERROR:shinysdr:An error occurred while executing the ShinySDR configuration file:
ERROR:shinysdr:  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 99, in _main_async
ERROR:shinysdr:    yield config_obj._wait_and_validate()
ERROR:shinysdr:ConfigException: No devices have been configured using config.devices.add(...).
INFO:shinysdr:Stopping factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7fa2c0216410>
INFO:shinysdr:Main loop terminated.
CRITICAL:shinysdr:Unhandled error in Deferred:
CRITICAL:shinysdr:
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 255, in _connect_to_device
    port=actual_port)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 137, in connect_to_rigctld
    proxy_ctor=_HamlibRig)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 168, in _connect_to_daemon
    protocol = yield connected
ConnectError: An error occurred while connecting: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionLost'>: Connection to the other side was lost in a non-clean fashion: Connection lost.
-----------------------------

mathisono, 2020-10-19:
OOOCRITICAL:shinysdr:Unhandled error in Deferred:
CRITICAL:shinysdr:
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/defer.py", line 654, in _runCallbacks
    current.result = callback(current.result, *args, **kw)
  File "/home/w6bb/shinysdr/conFiG/config.py", line 30, in <lambda>
    d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, shinysdr.devices.AudioDevice('Built-in Input', sample_rate=48000, channel_mapping='IQ')))
NameError: global name 'shinysdr' is not defined

kpreid, 2020-10-19:
Whoops, the manual's config example is a little inconsistent with naming. Add
```
from shinysdr.devices import AudioDevice
```
at the top if it's not already present, then change the `add` line like this, changing `shinysdr.devices.AudioDevice` to just `AudioDevice`.
...
d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, AudioDevice('Built-in Input', sample_rate=48000, channel_mapping='IQ')))
```

mathisono, 2020-10-19:
That was progress .... one must now define a proper audio device!!! it should be something like this "plughw:0,0"
""
gr::log :INFO: audio source - Audio source arch: alsa
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM Built-in Input
gr::log :ERROR: audio_alsa_source0 - [Built-in Input]: No such file or directory
""
""
d = hamlib.connect_to_rig(config.reactor, options=['-m', '2519', '-r', '/dev/ttyUSB0', '-s', '57600'])
d.addCallback(lambda rig: config.devices.add('RSHFiq',rig, AudioDevice('plughw:0,0', sample_rate=48000, channel_mapping='IQ')))
config.wait_for(5)
""

mathisono, 2020-10-19:
still no joy....

```
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

INFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f06aeb77460>
INFO:shinysdr:[unfiltered]
ERROR:shinysdr:An error occurred while executing the ShinySDR configuration file:
ERROR:shinysdr:  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/main.py", line 99, in _main_async
ERROR:shinysdr:    yield config_obj._wait_and_validate()
ERROR:shinysdr:ConfigException: No devices have been configured using config.devices.add(...).
INFO:shinysdr:Stopping factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f06aeb77460>
INFO:shinysdr:Main loop terminated.
rig_open: error = IO error 
CRITICAL:shinysdr:Unhandled error in Deferred:
CRITICAL:shinysdr:
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 255, in _connect_to_device
    port=actual_port)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 137, in connect_to_rigctld
    proxy_ctor=_HamlibRig)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/plugins/hamlib/__init__.py", line 168, in _connect_to_daemon
    protocol = yield connected
ConnectError: An error occurred while connecting: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionLost'>: Connection to the other side was lost in a non-clean fashion: Connection lost.
].
```

kpreid, 2020-10-20:
You need `config.wait_for(d)`, not `config.wait_for(5)`. This specifies waiting for `d` to finish its work (including adding the audio device as specified in addCallback). It does not take a time delay.

mathisono, 2020-10-20:
It works... Then the Web interface freezes becoming unresponsive, and there are no demod plugins available.  IDK?  Something is crashing on the backend of the server!!!! The server even keeps running after ctlC.

kpreid, 2020-10-20:
Unfortunately a problem like that generally indicates some issue with the GNU Radio components being used, and I cannot suggest any simple troubleshooting steps for you.

When you say no demod plugins at all, where exactly are you reading that and what does it say? If in the log, show me the text. If in the UI, a screenshot.

It's expected that various plugins will be only available if additional software is installed (e.g. rtl_433), but if the basic modes like AM, FM, etc. are missing then something's gone wrong with the installation of ShinySDR itself.

mathisono, 2020-10-20:
I have all of the plugins I need! 

I dont know how the backed was put together to handle an IQ audio. It broken according to what Im doing. Aside from the confusion about configuration. I will stick to running rtl SDR's and HackRF's for now.  

Let just say that as soon as I select the audioIQ device and click on the spectrum display to add another receiver, the web interface freezes. I've captures the terminal after the crash... and linked to screen captures.
------
https://drive.google.com/file/d/158adg1rRcIvbNPkj8pBQyeavy5HNTVXk/view?usp=sharing
------
https://drive.google.com/file/d/1plklrXFj_fGgdcYMaQKCZwbtSMEdsoL4/view?usp=sharing
------

term from start to crash....
``` 
shinysdr ~/shinysdr/conFiG
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2018.01.1 
INFO:shinysdr:3625000000.0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
OOOFound Rafael Micro R820T tuner
OOOOOOOOOOOOOO[R82XX] PLL not locked!
OOOOOOOOOOO[R82XX] PLL not locked!
OOOOOOINFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f5611672d70>
INFO:shinysdr:Stopping factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f5611672d70>
OOOOOOINFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f561168d0a0>
OOERROR:shinysdr:Unrecognized level description from rigctld: u''
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Constructing...
OOOOgr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: Switching RF device to osmo2
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: ...done reconnecting (6.47306442261 ms).
INFO:shinysdr:Restoring state...
OOOOOOOgr::log :INFO: audio source - Audio source arch: alsa
Ogr::log :INFO: audio source - Audio source arch: alsa
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Loading mode plugins...
INFO:shinysdr:Mode RTTY unavailable
No module named radioteletype.demodulators
OOOOOOINFO:shinysdr:Mode PSK31 unavailable
No module named radioteletype.demodulators
OINFO:shinysdr:Mode DSD unavailable
dsd.dsd_block_ff not found.
INFO:shinysdr:...done mode plugins.
Ogr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: Rebuilding connections because: added receiver a
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: ...done reconnecting (1.12414360046 ms).
INFO:shinysdr:Starting web server...
OINFO:shinysdr:SiteWithDefaultHeaders starting on 8100
INFO:shinysdr:Starting factory <shinysdr.i.network.base.SiteWithDefaultHeaders instance at 0x7f55fdb1c870>
INFO:shinysdr:WebSocketDispatcherProtocol (FactoryWithArgs) (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7f55fdb1c0a0>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f55fdb1c730>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/l-p-i9CzcZ52ihUsjAap2C/
OOOOOOOOOOOOOOOINFO:shinysdr:Starting RFC 6455 conversation
OINFO:shinysdr:Stream connection to /l-p-i9CzcZ52ihUsjAap2C/audio-stream?rate=48000
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio callback
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: ...done reconnecting (4.71305847168 ms).
gr::log :INFO: audio source - Audio source arch: alsa
OINFO:shinysdr:Starting RFC 6455 conversation
OINFO:shinysdr:Stream connection to /l-p-i9CzcZ52ihUsjAap2C/radio
[... 167 lines cut ...]
--- <exception caught here> ---
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/internet/tcp.py", line 327, in connectionLost
    protocol.connectionLost(reason)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/protocols/policies.py", line 125, in connectionLost
    self.wrappedProtocol.connectionLost(reason)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 417, in connectionLost
    self.inner.connectionLost(reason)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 332, in connectionLost
    self.__audio_source.remove_audio_callback(self.__callback)
[... cut ...] e_audio_callback
    return self.__receive_flowgraph.remove_audio_callback(callback)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/top.py", line 214, in remove_audio_callback
    self._do_connect()
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/top.py", line 226, in _do_connect
    raise Exception('reentrant reconnect or _do_connect crashed')
exceptions.Exception: reentrant reconnect or _do_connect crashed
```

mathisono, 2020-10-20:
I went ahead and check out the direct audio device config. The waterfall is intermittent with a 2sec delay, and if you click anything the web UI becomes unresponsive. 

```
#config.devices.add(u'rig', AudioDevice(rx_device='plughw:0,0', sample_rate=8000, channel_mapping=1, usable_bandwidth=(0, 8000)))
```

kpreid, 2020-10-21:
`gr::log :ERROR: audio_alsa_source1 - [plughw:0,0]: Device or resource busy`

Based on this log line, it looks like your audio devices don't cooperate with being reopened promptly. If you look at shinysdr/devices.py, around line 575 you should find:

```
    # implement IRXDriver
    def notify_reconnecting_or_restarting(self):
        # Under some conditions on Mac, gnuradio.audio.source may stop working when the flowgraph is modified. Therefore, recreate it, which causes a glitch but doesn't leave the device permanently nonfunctional.
        self.__init_source()
```

Replace `self.__init_source()` with `pass` and see if it works better. That will disable reopening the audio device whenever a change is made.

mathisono, 2020-10-21:
That fixes the waterfall. Adding a receiver still crashes the backend.

```
$ shinysdr ~/shinysdr/conFiG
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2018.01.1 
INFO:shinysdr:3625000000.0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
OOOFound Rafael Micro R820T tuner
OOOOOOOOOOOOOO[R82XX] PLL not locked!
OOOOOOOOOO[R82XX] PLL not locked!
OOOOOOOINFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f8cb4f1dd20>
INFO:shinysdr:Stopping factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f8cb4f1dd20>
OOOOOOINFO:shinysdr:Starting factory <shinysdr.plugins.hamlib._HamlibClientFactory instance at 0x7f8cb4f38050>
OOERROR:shinysdr:Unrecognized level description from rigctld: u''
gr::log :INFO: audio source - Audio source arch: alsa
OINFO:shinysdr:Constructing...
OOOOgr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: Switching RF device to osmo2
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: Rebuilding connections because: initialization, switched device
gr::log :INFO: audio source - Audio source arch: alsa
INFO:shinysdr:Flow graph: ...done reconnecting (7.48705863953 ms).
INFO:shinysdr:Restoring state...
WARNING:shinysdr:Discarding nonexistent state <shinysdr.i.session.AppRoot object at 0x7f8cb4f562d0>.source_name = osmo2
INFO:shinysdr:Starting web server...
OOOOOOOINFO:shinysdr:Loading mode plugins...
INFO:shinysdr:Mode RTTY unavailable
No module named radioteletype.demodulators
OINFO:shinysdr:Mode PSK31 unavailable
No module named radioteletype.demodulators
OINFO:shinysdr:Mode DSD unavailable
dsd.dsd_block_ff not found.
INFO:shinysdr:...done mode plugins.
INFO:shinysdr:SiteWithDefaultHeaders starting on 8100
INFO:shinysdr:Starting factory <shinysdr.i.network.base.SiteWithDefaultHeaders instance at 0x7f8ca1418370>
INFO:shinysdr:WebSocketDispatcherProtocol (FactoryWithArgs) (WebSocketFactory) starting on 8101
INFO:shinysdr:Starting factory <shinysdr.twisted_ext.FactoryWithArgs instance at 0x7f8ca1418050>
INFO:shinysdr:Starting factory <txws.WebSocketFactory instance at 0x7f8ca1418410>
INFO:shinysdr:ShinySDR is ready.
INFO:shinysdr:Visit http://localhost:8100/l-p-i9CzcZ52ihUsjAap2C/
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOINFO:shinysdr:Starting RFC 6455 conversation
OINFO:shinysdr:Stream connection to /l-p-i9CzcZ52ihUsjAap2C/radio
OOOOgr::log :INFO: audio source - Audio source arch: alsa
OOOOOOOOINFO:shinysdr:Starting RFC 6455 conversation
OINFO:shinysdr:Stream connection to /l-p-i9CzcZ52ihUsjAap2C/audio-stream?rate=48000
INFO:shinysdr:Flow graph: Rebuilding connections because: added audio callback
gr::log :INFO: audio source - Audio source arch: alsa
OOOOINFO:shinysdr:Flow graph: ...done reconnecting (70.2359676361 ms).
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOINFO:shinysdr:Flow graph: Switching RF device to RSHFiq
Ogr::log :INFO: audio source - Audio source arch: alsa
OOOOOINFO:shinysdr:Flow graph: Rebuilding connections because: switched device
gr::log :INFO: audio source - Audio source arch: alsa
OOOOOOINFO:shinysdr:Flow graph: ...done reconnecting (199.266910553 ms).
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOgr::log :INFO: audio source - Audio source arch: alsa
gr::log :ERROR: audio_alsa_source1 - [plughw:0,0]: Device or resource busy
CRITICAL:shinysdr:
Traceback (most recent call last):
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/web/server.py", line 199, in process
    self.render(resrc)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/web/server.py", line 259, in render
    body = resrc.render(self)
  File "/home/w6bb/.local/lib/python2.7/site-packages/twisted/web/resource.py", line 250, in render
    return m(request)
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_http.py", line 120, in render_POST
[... 62 lines cut ...]
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/i/network/export_ws.py", line 332, in connectionLost
    self.__audio_source.remove_audio_callback(self.__callback)
[... cut ...]
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/devices.py", line 575, in notify_reconnecting_or_restarting
    self.__init_source()
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/devices.py", line 538, in init_source
    ok_to_block=True)
  File "/usr/lib/python2.7/dist-packages/gnuradio/audio/audio_swig.py", line 154, in make
    return _audio_swig.source_make(*args, **kwargs)
exceptions.RuntimeError: audio_alsa_source

OOO

```

kpreid, 2020-10-21:
```
  File "/home/w6bb/.local/lib/python2.7/site-packages/ShinySDR-0.0.0-py2.7.egg/shinysdr/devices.py", line 575, in notify_reconnecting_or_restarting
    self.__init_source()
```
This line in the traceback indicates that the change I specified was not applied. Perhaps you removed the line under `__init__` and not the one in `notify_reconnecting_or_restarting`? Or, actually, I see you have it in site-packages — if you edited the downloaded copy then you need to either rerun `setup.py install` or edit the copy that's in the site-packages directory.

(In any case, this problem indicates that I really ought to disable this by default on Linux, if not everywhere. But that won't happen today.)