kpreid/shinysdr #150: enable antenna port power...

mathisono, 2020-11-02:
I'm wondering where I might place the ```-p 1``` argument for the hackRF to activate power on its antenna port for a preAmp?

kpreid, 2020-11-02:
Options like that are parameters to the osmosdr device string, but not spelled the same way as the hackrf command line tools. My config file for HackRF with power looks like:

```
config.sources.add('hackrf', OsmoSDRDevice('hackrf=,bias=1', sample_rate=20e6, correction_ppm=19))
```

(Note the comma, and no space.)

This and other options are documented at the [gr-osmosdr wiki](https://osmocom.org/projects/gr-osmosdr/wiki#HackRF-Source-Sink), and the final word of course is [the source code, `gr-osmosdr/lib/hackrf/hackrf_source_c.cc`](https://git.osmocom.org/gr-osmosdr/tree/lib/hackrf/hackrf_source_c.cc).

mathisono, 2020-11-02:
```config.devices.add(u'osmo1', OsmoSDRDevice('hackrf=4f4397,bias=1', name='hackrf--(APN-450)'),PositionedDevice(xyz, xyz))```

Interesting that you apply your ppm there! my SDR farm is at a remote site temperature really plays a big part in my ppm drift so much so I've started putting fans on the RTL v3, they get really HOT!!!

kpreid, 2020-11-02:
I assume you've got the bias power working!

closed 2020-11-02 by kpreid

mathisono, 2020-11-02:
yes