kpreid/shinysdr #151: import setuptools ImportError: No module named setuptools

OwensDataCenter, 2021-03-06:
import setuptools
ImportError: No module named setuptools

kpreid, 2021-03-06:
Apparently your Python doesn't have setuptools already installed, so you will need to install it.

If you are on Linux, look for a package for it in your distro's package manager. Otherwise, see https://packaging.python.org/tutorials/installing-packages/ for some advice on how to get setuptools installed.

OwensDataCenter, 2021-03-06:
Windows 8.1 (For Testing Files From GitHub), Windows 10 (Main Server For Website)

OwensDataCenter, 2021-03-06:
I am a newbie on this...
I am tired of getting errors this week.
Please help me.

kpreid, 2021-03-06:
I'm sorry, I don't have any information about Python setuptools on Windows. Please consult the link I provided. Perhaps someone else will have recommendations too.

OwensDataCenter, 2021-03-07:
Ok, Thank You

closed 2021-03-20 by kpreid