kpreid/shinysdr #152: Audio spectrum and oscilloscope questions

NeuroForLunch, 2021-05-01:
First off, amazing project! 👍 

Now the questions:
I want to use the audio spectrum for frequencies above 18.5k so how can I make this settable? I need it to go to at least 196kHz and preferably beyond so I can measure ultrasound. 

How can I set the oscilloscope to only view a part of the audio spectrum. Say, 16k to 18k. 

If these actions are not currently possible, can you let me know where to start building them? And should I just start on the Python3 branch?

kpreid, 2021-05-01:
> I want to use the audio spectrum for frequencies above 18.5k so how can I make this settable? I need it to go to at least 196kHz and preferably beyond so I can measure ultrasound.

You're referring to the [standalone audio spectrum analyzer](https://shinysdr.switchb.org/tools/audio-spectrum), right, not the ShinySDR server? In that case, the sample rate is entirely dependent on your audio device's capabilities and default settings. (For example, if you're using macOS, you can change the default sample rate using the “Audio MIDI Setup” tool; I'm not familiar with the Windows and Linux equivalents.)

If you're using the ShinySDR server, then in your config file, pass a `sample_rate` option to the `AudioDevice` constructor. (This will not help if the hardware does not support the requested rate, or it may succeed but not actually show you any higher-frequency signals, depending on how your OS's audio subsystem chooses to handle the request.)

> How can I set the oscilloscope to only view a part of the audio spectrum. Say, 16k to 18k.

The _oscilloscope_ display does not do any frequency selection. For the spectrum analyzer, scrolling vertically, or using two fingers on a touchscreen, will allow you to zoom in to any frequency range.

> If these actions are not currently possible, can you let me know where to start building them? And should I just start on the Python3 branch?

All of the spectrum display code is in JavaScript so the Python version issues don't affect it. [spectrum.js](https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/client/widgets/spectrum.js) contains the graphics and UI elements, [client-source.js](https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/client/audio/client-source.js) is the part that accesses the microphone or other audio device, and [audio-spectrum-main.js](https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/tools/audio-spectrum-main.js) is the top-level code for the spectrum analyzer page.

NeuroForLunch, 2021-05-01:
I am using macOS and yes I'm referring to the standalone audio spectrum analyzer. I have an external mic that I can connect via the midi settings that has a sample rate of 384 kHz and can record ultrasonic frequencies, however, the 18.5K limit remains.

Also, I need to analyze certain compression-wave (not radio wave) frequencies because encoded data is being sent via FM and some other modulation. Any clues on how to do this?

kpreid, 2021-05-01:
> however, the 18.5K limit remains.

Try this: in `audio-spectrum-main.js`, change the line

```javascript
 const audioContext = new AudioContext();
```

to

```javascript
  const audioContext = new AudioContext({
    sampleRate: 384000,
  });
```

It would require more modifications to automatically select the best available sample rate, but this should work for your purpose.

---

> Also, I need to analyze certain compression-wave (not radio wave) frequencies because encoded data is being sent via FM and some other modulation. Any clues on how to do this?

The standalone spectrum analyzer page does not have any demodulation functionality. You would have to use the ShinySDR server to do that. It does not have many features useful for analyzing a signal, however — the most it could do, unless you write a custom demodulator plugin, is demodulate the FM and provide that as an audio signal.

For your purposes, you might be better off using GNU Radio Companion instead of ShinySDR, to assemble an audio device source, FM demodulator and other processing blocks; display the resulting waveforms; and modify the setup as needed to understand the signal.

NeuroForLunch, 2021-05-01:
I appreciate your advice.

closed 2021-05-01 by NeuroForLunch