kpreid/shinysdr #155: Can we (I) have frequency offset added ?

zanco, 2021-08-22:
Running shinysdr with rtl device, ppm set, but running behind a satellite LNB with 9.750 GHz Local Oscillator 

Would be nice if I could add this offset and have the direct frequency on my screen without having windows calculator open to do the math for me :-)

Thanks,

Ben - PE2BZ from the Netherlands

kpreid, 2021-08-22:
Combine `shinysdr.devices.FrequencyShift(-9.750e9)` with your RTL device in the config file.

This is documented in [the "special-purpose devices" section of the manual](https://shinysdr.switchb.org/manual/configuration#special).

zanco, 2021-08-23:
Thank for the explanation !

The frequency display(ed) is not ready for frequencies over 9.9 GHz is that right ?

kpreid, 2021-08-23:
It shouldn't be hard to change.

The maximum digits are set here in this loop:  https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/client/widgets/basic.js#L622

The font size will probably have to be reduced too: https://github.com/kpreid/shinysdr/blob/master/shinysdr/i/webstatic/client/ui.css#L513

closed 2021-08-28 by kpreid