kpreid/shinysdr #156: Adding transmit capability

alphafox02, 2021-12-28:
I’ve been using the pull request version for python3 for some time now, on and off. I wondered if there was still hope to see a transmit capability added as mentioned briefly in the documentation. 

Hope all is well.

kpreid, 2021-12-28:
Unfortunately, my personal interest in working on the project has waned. I may get back to it at some point, but don't expect a quick resumption.

Additionally, GNU Radio's architecture and the typical capabilities of source/sink driver blocks are not well suited to usably low-latency transmit-receive switching — and has often given me trouble even trying to implement ShinySDR's core feature of having multiple demodulators that can be added/removed/changed. If I get back to the project, it is likely that it would involve a complete rewrite of the server side, and no longer using Python. I've given some thought to what it would look like, but I can't promise a schedule.

(All that said, it'd be good to have an official Python 3-compatible version, and that's less of a problem than adding major new features, so I might get around to that sooner.)

alphafox02, 2021-12-28:
No rush, completely understand. I’ve included the python3 version in DragonOS and just got back around to trying it out today using an Intel NUC at a remote location. I was able to hear audio on a p25 phase 1 channel and could see gr-DSD doing its thing in a terminal. So that’s good. I was also able to use mode S so gr-air-modes is working. I never understood why I don’t see aircraft on the map though, assuming because there’s no gps info in the aircraft messages. 

Call me crazy, but I like when an application uses various tools. I do like hearing maybe at some point you might even redo things. Thanks again for putting in the time to get this to where it’s at now. I was thinking I’d try and do up a video on what I sorta know about the application.

Sorry for getting off track. If there a better place for continued discussion I’m all ears.

kpreid, 2021-12-28:
No problem. There isn't a separate discussion forum for ShinySDR.

I'm going to close this issue in favor of #124 since that is precisely about transmit mode support (and already covers some of the architecture questions in implementing it).

closed 2021-12-28 by kpreid

alphafox02, 2021-12-28:
Excellent, I’ll read up on the topic there. Have a great New Years!